teachop/FlexCAN_Library #11: Automotive OBDII CAN ports are already terminated.

Mike4U, 2016-02-12:
Guessing many may want to use this to look at their auto's OBD port.

The existing OBD CAN port under your car's dash is on a bus that is already terminated on both ends in order to run properly when nothing is plugged in.  Adding more termination resistors most likely will make matters worse when plugging into the port.

Mike4U, 2016-02-12:
Forgot to add the above refers to the termination statement in the readme.

teachop, 2016-02-12:
Yes, that is correct.  Termination resistors are located at both ends of the signal bus.

closed 2016-02-12 by teachop

Mike4U, 2016-02-13:
Newbie's don't know that.  Why don't you add it to the read me instead of:

 "Note that CAN will normally not work without termination resistors."

thanks

teachop, 2016-02-13:
Ok.  I will get some explanation of the proper termination concept.