teachop/FlexCAN_Library #13: MCP2551 and MCP2515 or just MCP2551?

antimodular, 2016-02-23:
Hi

I was wondering if you quickly could clarify the hook of the MCP2551 to the Teensy.

The Arduino libraries i have found seem to work on SPI and need MCP2551 and MC2515 together. As per this schematic: http://cdn.sparkfun.com/datasheets/Dev/Arduino/Shields/SparkFun_CAN-Bus_Shield_v13a.pdf

Does you library only need the MCP2551and needs it's CANL and CANH pins connected to the teensy?

Thanks,
Stephan

riodda, 2016-02-23:
you will need a transcriever between the teensy and canh canl lines

antimodular, 2016-02-23:
do you have a suggestion which transceiver chip i should use or a link to a sample schematic?
the MC2515 on the spark fun board is a SPI chip, so not the right thing to use.

thanks.

riodda, 2016-02-23:
i use ti svn chip like 
http://www.dx.com/p/bus-transceiver-sn65hvd230-can-communication-module-blue-385988#.Vsys8cvUUqI

you will have to follow this schematic for wiring

https://upload.wikimedia.org/wikipedia/commons/9/9e/CAN-Bus_Elektrische_Zweidrahtleitung.svg

antimodular, 2016-02-23:
oh i see.
i am already using the MCP2551 (which you refer to as the transceiver). so i am good to go i guess.
i just wanted to make sure i am not missing some other electronics. 

my goal is to have multiple teensy's talk to each other via the CAN transceivers.

[image]

riodda, 2016-02-23:
yes, you are got to go, can rx and tx will go to teensy can pins, tx to tx rx to rx, just remember about the termination of the bus

antimodular, 2016-02-23:
great.

one last question.
the CanTest example that comes with the github download, should i upload that to both teensys or would the echo node need a different sketch?

thank so much for your help.

riodda, 2016-02-23:
chanche just line 85 to avoid the 2 nodes send on the same id

antimodular, 2016-02-24:
great thanks again.
i ended up using this receiver and sender test code:
http://www.mikrocontroller.net/topic/370955#4189341

closed 2016-02-26 by antimodular