teachop/FlexCAN_Library #18: How to get raw data?

frmdstryr, 2016-08-15:
Is it possible to get the raw data? 

I just want to read the CAN bus pass it directly to the serial port for a pcap log.

Thanks!

riodda, 2016-08-16:
If you look at the example it will only print byte 0 of the received message, you simply modify in this way:

  if ( !rxTimer ) {
    while ( CANbus.read(rxmsg) ) {
      Serial.write(rxmsg.msg.id);
      Serial.write("-");
      Serial.write(rxmsg.buf[0]);
      Serial.write("-");
      Serial.write(rxmsg.buf[1]);
      Serial.write("-");
      Serial.write(rxmsg.buf[2]);
      Serial.write("-");
      Serial.write(rxmsg.buf[3]);
      Serial.write("-");
      Serial.write(rxmsg.buf[4]);
      Serial.write("-");
      Serial.write(rxmsg.buf[5]);
      Serial.write("-");
      Serial.write(rxmsg.buf[6]);
      Serial.write("-");
      Serial.writeln(rxmsg.buf[7]);
      rxCount++;
    }

frmdstryr, 2016-08-16:
Thanks for the prompt reply. I tried that but it's missing the initial flags, length, and i think a timestamp. 

The full can frame (flags + id + len + timestamp+ data)  is needed so it can be decoded properly in wireshark.   See:

[image]

The first 3 bits are the 3 flags. The bytes 0-3 (excluding the flag bits) is the id. The byte 4 is the length. Bytes 5-7 are the timestamp (i think) and bytes 8-(8+len) the rest is the data. 

It can be rebuilt if all the flags and timestamp are available (RTR and Error flag are missing).

Was just wondering if there's an easy way to do this, if not I'll attempt to make one

riodda, 2016-08-16:
I had a quick look to the code, in the msg  structure there are only this fields:

`typedef struct CAN_message_t {
  uint32_t id; // can identifier
  uint8_t ext; // identifier is extended
  uint8_t len; // length of data
  uint16_t timeout; // milliseconds, zero will disable waiting
  uint8_t buf[8];
} CAN_message_t;`

I think what you need has to be implemented.

frmdstryr, 2016-08-28:
Thanks. Made some modifications to add the missing flags. Working great now.

closed 2016-08-28 by frmdstryr

riodda, 2016-08-29:
cool, please make a pull request!