teachop/FlexCAN_Library #20: No indication of full buffers

riodda, 2016-09-11:
i had a very odd behaviour of the library that drove me a bit creasy for a couple of hours (even if is not a real bug).
I guess that the micro has 8 fuffers to send messages and when those are full there is no indication that the message you send in not being sent.
Here below is part of the code that gave me problem (it's a GPS that sends data trough canbus):

First is with no delay and the 9th frame in not sent (don't show up in PCANview)

`  if (CAN[0].en)
      CANbus.write(can_pos);  
  if (CAN[1].en)    
      CANbus.write(can_pos_fil);
  if (CAN[2].en)
      CANbus.write(can_nav); 
  if (CAN[3].en)
      CANbus.write(can_nav_fil);
  if (CAN[4].en)
      CANbus.write(can_dof1);
  if (CAN[5].en)
      CANbus.write(can_lap); 
  if (CAN[6].en)   
      CANbus.write(can_dof2);  
  if (CAN[7].en)   
      CANbus.write(can_gyr); 
  if (CAN[8].en)   
      CANbus.write(can_dop);   `

As soon as i add the delay just before the last CANbus.write the 9th frame appears and i can see it in the PCANview.

`  if (CAN[0].en)
      CANbus.write(can_pos);  
  if (CAN[1].en)    
      CANbus.write(can_pos_fil);
  if (CAN[2].en)
      CANbus.write(can_nav); 
  if (CAN[3].en)
      CANbus.write(can_nav_fil);
  if (CAN[4].en)
      CANbus.write(can_dof1);
  if (CAN[5].en)
      CANbus.write(can_lap); 
  if (CAN[6].en)   
      CANbus.write(can_dof2);  
  if (CAN[7].en)   
      CANbus.write(can_gyr); 
  delay(1);  // Added delay to empty the can send buffers
  if (CAN[8].en)   
      CANbus.write(can_dop);   `

i think that a flag to war if all the buffers are full should be added.

pawelsky, 2016-09-11:
There already is such a 'flag' - check the value returned by the write function.

riodda, 2016-09-11:
Thanks ! My fault, i was not looking for it, i haven't found it in the example i started with long time ago and i thought it was not implemented.