teachop/FlexCAN_Library #25: Special Baud speed

roblesterjr04, 2016-12-19:
High, this is probably asking too much. I don't know enough to modify your code with any level of success, So I was wondering if you could make a small change. I'm not unfamiliar with this stuff, but obviously you're working below the abstraction and I haven't gotten that good yet. I simply need a custom baud rate added. Would you be able to include the necessary changes for a baud speed of 47.619 kbp/s? I'm trying to interface with a network of devices that supposedly run at that speed and I'm just not sure where my failure is. If there are any other configuration values you need, please let me know and I think i can find them in the information i've gathered so far. Thank you.

collin80, 2016-12-20:
It seems it should be possible. It appears that the CAN bus clock is set to 16MHz and 16Mhz / 47619 = 336.0003 which is close enough to 336 for government work. 336 should be able to be divided by 16 to get 21 and 21 should work for setting up the bit timing registers. So, I think this request would be possible. Now, will it get added to the library? I don't know. I'm thinking of adding the ability to set arbitrary baud rates to my fork of the library. Then there would be no need to ask for special rates.

collin80, 2016-12-21:
I just added support for generating arbitrary baud rates to my fork at https://github.com/collin80/FlexCAN_Library which is a fork of a fork of this library. So, you can use mine or just lift the setup code from mine and put it back into this version of the library and it should still work. As it turns out, my auto baud generator uses different numbers than my hand calculated version above but the result should be the same.