teachop/FlexCAN_Library #27: Can message looping on Teensy 3.6

Ligozila, 2017-02-02:
Good morning,

I am new to the CAN community and was trying on a project with testing the CAN bus line. 

I have a Teensy 3.6 in hand. With Teensy having 2 Can line available, I am trying to send a message from Can0, receive from Can1 and compare the message. I still did not receive the CAN Transceivers module yet, and hopefully they will arrive soon. However, to test out the code, I have solder 1k resistors between the Can0_TX and Can1_RX, Can0_RX and Can1_TX, I hope this will emulate the signal from the CAN Transceivers in-between.

I have complied successfully and download the sketch "CANTest.ino" into Teensy, I have measured and got some kind of pulse signals from the Can0_TX and Can1_RX. Signal generated from Can0_TX and reach to Can1_RX via the 1k resistor. 

Problem: 
It seems Can1_RX never reads the signal, Can1.available() always return 0.

Does the resistor linking between the Can TX and RX is valid? Is there some signal conditions, like pulling the line down or high to signal the line is available? do I need to make change to the sketch? or should I add another Teensy, so 1 transmit and another receive... etc 

It will be great to have your advise and thought!

pawelsky, 2017-02-03:
> Does the resistor linking between the Can TX and RX is valid?

No it is not. Use proper CAN transceivers or at least http://www.cnblogs.com/shangdawei/p/4714957.html

Ligozila, 2017-02-06:
Good Morning, 
Hope your guys have a happy long weekend. (It is a long weekend in NZ)

Thank you for the reply. I was wondering does anyone done something similar, testing the code when there is not transceiver available in hand.

However, I have found out more over the weekend. And this maybe interested to you. The way I linking of resistors are definitely wrong. Because the nature of CAN bus, Can_L is the inverted signal of Can_H, and the outgoing & incoming Can message are appear on the same lines, then whatever that transmitted on the TX will be loop back on the RX line (because both TX and RX are sharing the same bus line). But whatever that received on both (Can_H & Can_L) line are only outputs on the RX, and will not be appear on the TX. In most of the Can enabled microcontroller, the RX and TX transmitting in the RX and TX are having an internal pull up resistor, which means, the RX, TX pin will be high impedance with signal 'High' and actively (electronically) pull low with signal 'Low'. This is very similar to the Can bus line, that, 'Recessive' is high impedance, and 'dominant' is actively pull high or low depending on the line.

Siemens have an application note for linking the Can bus RX and TX without the transceiver (attached below). In there, it seems since the Can_L and Can_H are 1 channel, therefore, the application note links both TX and RX in 1 line as well. And the TX and RX are linked with a diode, with this setup, the TX signal will be visible from RX, but the RX signal (from the other TX line!) will be isolated. From the problem I experienced above (the message is not picking up!), without the Ack in the line, or maybe without the loopback on the RX, the hardware level code will keep sending out the message until time_out is reached.  
[siemens_AP2921.pdf](https://github.com/teachop/FlexCAN_Library/files/755660/siemens_AP2921.pdf)

The application note did comment that, without the transceiver, the total **max line distance should not be over 1 Meter,** (physical electrical connection) some comment did say that keep it under 10cm before the noise overrun the signal.

I will implement Siemens recommended setup and write a bit more on the result. Sorry, I keep you read for a long paragraph, but did not have any real life experience to confirm the situation yet.

Ligozila, 2017-02-06:
Good Morning,
Hope your guys have a happy long weekend. (It is a long weekend in NZ)

Thank you pawelsky for the reply. I was wondering does anyone done something similar, testing the code when there is not transceiver available in hand.

It is very useful, especially the setup for the TX and RX line.

Ligozila, 2017-02-09:
The Pin setup works great. thank you for the great help

closed 2017-02-09 by Ligozila