teachop/FlexCAN_Library #29: ReadMe.md Lines 6 & 14 Connecting Teensy to Transceiver

CNCQuilter, 2017-09-19:
Line 6 & 14 of the Updated_ReadMe.md file seem to contain conflicting information regarding connecting a Teensy to a CAN Transceiver.

Should the TX and RX Teensy pins be connected to the Transceiver TXD & RXD pins respectively (line 6) or just the reverse (line 14)?

tonton81, 2018-08-18:
TX to TX, RX to RX. Just like UART, you won't blow up anything if they're backwards, it just won't work.

pawelsky, 2018-08-19:
Well, actually that answer (especially the "just like UART" part) may make things even more confusingas with UART you usually connect TX to RX and vice versa. 

In case of the CAN transceiver the situation is a bit different mainly due to different (and a bit confusing) naming convention used. Transciever's TXD is a Transmit Data **Input** (so actually a receive pin) while RXD is Receive Data **Output** (so actually a transmit pin). 

That is the reason why one should connect Teensy's TX pin with transciever's TXD pin and RX with RXD.