teachop/FlexCAN_Library #32: Enable pins of CAN bus is reversed

wt70707, 2019-01-30:
//if using enable pins on a transceiver they need to be set on
  pinMode(2, OUTPUT);
  pinMode(35, OUTPUT);

  digitalWrite(2, HIGH);
  digitalWrite(35, HIGH);

pawelsky, 2019-01-30:
Cool, but you are reporting the issue to the wrong repository as this library does not deal with enable pins.

You probably meant this one https://github.com/collin80/FlexCAN_Library

wt70707, 2019-01-30:
Thanks Pawelsky, there is no place to report issues at collin80. And all the examples here also have the above code to do the activation.

pawelsky, 2019-01-30:
First of all the fact thet there is no place to report issues in Collin's repository does not mean you should report in to other random repository. 

Secondly, no, there are no examples here that have the mentioned code.

Finally, even if there were this is not even a bug in the library, as the transcievers that you use do not have the ENABLE pins. They have Rs pins which for this particular chip (SN65HVD230D) enables the High-speed mode when pulled low, low-power mode when pulled high, and manages slope control when pulled down with a 10kohm-100kohm resistor.

If you really want to argue more contact colin80 directly (e.g. via the pjrc forum: https://forum.pjrc.com/members/44583-CollinK) or create a pull request in **HIS** repo.

collin80, 2019-01-30:
Sorry for the mix up involving my repo. I have fixed the problem and if you have trouble you can now directly create an issue and yell at me for it.

wt70707, 2019-01-30:
Sorry Pawelsky. I am asking question not reporting a bug of the library. Your explanation make many sense. I will check the spec of SN65HVD230D. As you clearly mentioned about the pull down to control the slope, do you think I should put one, as I got very unstable CAN bus reading. 
I will put my question on Collin80.